dalathegreat/Battery-Emulator #1228: [BUG] Board Reset - Panic / Exception

demon1300, 2025-06-11:
**Describe the bug**
Flashed v8.14.0 but am receiving a CPU panic/exception causing a reboot.  Noticed straight away afterflashing that the GUI was very laggy and within a few seconds, was given a yellow status.  Captured a log via serial to help debugging which is attached.

Of note, my Atto-3 battery is not crashed - so only other modification is commenting out the estimated SOC in this version so that I could switch modes via the GUI if needed using the new feature (why i was updating in the first place).

The same user settings for version 8.13.0 work fine - nothing was changed as far as user settings between versions.

**To Reproduce**
Upload to board v8.14.0 and within a few seconds of the reboot, the hardware resets.

**Expected behavior**
After flashing the BE should operate as usual.

**Screenshots**

[image]

**Version and settings (please complete the following information):**
- Please attach your USER_SETTINGS files for easier debugging
- Software version: 8.0.14
- Battery used: BYD-ATTO3
- Inverter communication protocol: BYD_MODBUS
- Hardware used for Battery-Emulator: HW_LILYGO
- CONTACTOR_CONTROL: N/A
- MQTT: Yes

[USER_SETTINGS.h.txt](https://github.com/user-attachments/files/20689852/USER_SETTINGS.h.txt)

[Lilygo_DebugLog.txt](https://github.com/user-attachments/files/20689823/Lilygo_DebugLog.txt)

dalathegreat, 2025-06-11:
Issue seems related to MQTT, I am unable to diagnose this one further without an MQTT broker setup

Cees-JanBerg, 2025-06-15:
i have the same problem. 
use MQTT IO, musqito broker on homeassistant. 

//MQTT credentials
#define MQTT_SERVER "192.168.2.35"  // MQTT server address
#define MQTT_PORT 1883                 // MQTT server port
#define MQTT_USER "*******"                   // MQTT username, leave blank for no authentication
#define MQTT_PASSWORD "*********"

BEMSman, 2025-06-15:
Is the #define MQTT_PASSWORD "**" an issue

mine looks like this for no authentication:

//MQTT credentials
#define MQTT_SERVER "10.10.10.254"  // MQTT server address
#define MQTT_PORT 1883                 // MQTT server port
#define MQTT_USER ""                   // MQTT username, leave blank for no authentication
#define MQTT_PASSWORD ""               // MQTT password, leave blank for no authentication

Cees-JanBerg, 2025-06-15:
i don't think so. i now use 8.13 with no problems.

dalathegreat, 2025-06-15:
@kyberias / @No-Signal  , sounds like either PR #1208 , #1154 or #1125 are prime suspects for this new crash

Morris3190, 2025-06-15:
Same symptoms but I am not running MQTT. Tried to increase MQTT_MSG_BUFFER_SIZE in case part of code would still be in use despite the //#define MQTT in the user_settings.h
So probably something else ...

kyberias, 2025-06-16:
> Same symptoms but I am not running MQTT. Tried to increase MQTT_MSG_BUFFER_SIZE in case part of code would still be in use despite the //#define MQTT in the user_settings.h, but no luck ... still getting panic resets. So probably something else ...

This is something else if MQTT is not used. Please report a separate issue with all the necessary defails.