dalathegreat/Battery-Emulator #1229: Tesla model 3 LFP 12v battery mot charging issue?

jmacker66, 2025-06-11:
HI, I have a Tesla model 3 LFP battery for 4 months now and it has been working fine until last week when the 12v battery stopped charging. I had measured the voltage before and it was 14.2 volts but now it looks like the output from the PCS is not active, which causes the lead acid battery to run flat after 2-3 days and the contactors drop out. I recharged the battery and restart and its fine until the battery runs flat again.
Nothing has changed to cause this and everything looks okay, but I need to resolve this if its possible. Any ideas?
Thanks.

jmacker66, 2025-06-22:
Hi, I cant be the only one with this issue? Is there any help available? I'm thinking of re-engaging the PCS cooling loop in an attempt to keep the PCS cool while running. The LilyGo is displaying a PCS ambient temp of  423.34oC which could not be correct? It must be a scaling issue on the unit and maybe means 42.33oC? Anyway I think the cooling 
loop will help...?

Webwanze, 2025-06-29:
I have hooked up a lilygo on a Tesla LR battery. Battery was still in the car and cooling lines were connected. Software version 8.9.0 flashed. 

It was reporting a PCS ambient temp of 231.20 DegC. Pack Temp Min was displayed with 22.50 DegC at this time.

solartech-fl, 2025-12-28:
I have this problem on two packs one is a lfp pack and the other one is a long range pack I have a total of 10 battery packs that I have tested. the way I got around it was just using a battery charger 5 amps minimum and it just maintains the battery off of 120 volt Outlet I have similar temperatures reported as well and under more battery info it says that the PCS is faulted and ridiculous temperatures listed