dalathegreat/Battery-Emulator #1239: [Feature request]

HansNetcom, 2025-06-15:
**Is your feature request related to a problem? 
I,m working on a configuration with Fronius invertor and CHADEMO-BATTERY. using a 2024 Nissan Leaf
If the BE boots-up it sends instantly this message sequence.

 0.082 Battery emulator 8.15.dev build Jun 15 2025 17:12:23
       0.086 Event: The board was reset via software, webserver or OTA. Normal operation
       0.180 Creating Access Point: Battery Emulator
       0.181 With password: 123456789
       0.181 Access Point created.
       0.181 IP address: 192.168.4.1
       0.191 Connecting to Wi-Fi...
       0.251 Event: Wifi connected.
       0.251 Wi-Fi connected. RSSI: 87 dBm, IP address: 0.0.0.0
       1.000 Event: Battery under minimum design voltage. Charge battery to prevent damage!
       1.001 Event: Battery is completely discharged
       1.326 Wi-Fi Got IP. IP address: 192.168.0.192
       1.610 MQTT initialized
       1.890 Event: MQTT connected.
       1.890 Publishing buttons discovery
       1.901 MQTT connected

Looks OK except that the CHADEMO plug is not (yet) inserted.
The BE sees this as an event 

 1.000 Event: Battery under minimum design voltage. Charge battery to prevent damage!
 1.001 Event: Battery is completely discharged

This is a wrong message since the BE could not yet read the battery.
When I connect the CHADEMO plug to the leaf the BE send a canmessage to the leaf with the wrong battery level so the leaf refuses to start charging/discharging

**Describe the solution you'd like**
It would be better to wait on a CHADEMO_CONNECTED signal and then start messaging about the battery.
Set a default basic Battery voltage and send this to the leaf to start successful handshaking for charging/discharging.

dalathegreat, 2025-06-15:
Great ideas @HansNetcom , I have implemented this in PR #1240

HansNetcom, 2025-06-17:
Thanks for the quick solution. Seems to work fine.
New challenges: 

1. The BE reports the wrong voltage to the leaf.
x200 minimum discharge voltage met or exceeded, stopping.

2.If the BE emulator is rebooting with the CHADEMO plug connected to the leaf, it gives this error.
Chademo plug connected to car.
Output after reboot
       2.409 MQTT initialized
       2.461 Charge Rate (kW): 0
       2.461 UNHANDLED STATE IN process_vehicle_charging_session()
       2.466 Event: MQTT connected.
       2.467 Publishing buttons discovery
       2.478 MQTT connected
       2.561 Charge Rate (kW): 0
       2.561 UNHANDLED STATE IN process_vehicle_charging_session()
       2.661 Charge Rate (kW): 0
       2.661 UNHANDLED STATE IN process_vehicle_charging_session()
       2.761 Charge Rate (kW): 0
       2.761 UNHANDLED STATE IN process_vehicle_charging_session()
       2.861 Charge Rate (kW): 0

dalathegreat, 2025-06-17:
Hi @HansNetcom 
I updated the PR, please re-try! direct download link: https://github.com/dalathegreat/Battery-Emulator/archive/refs/heads/bugfix/chademo-startup.zip

PR #1240 (2025-06-26): Chademo: Improve startup to throw less events
### What
This PR improves the Chademo start, to generate less false-positive events

### Why
Fixes #1239 

### How
We only update voltage after car has started communicating. We also init SOC% to 50%, as on other integrations to avoid fake battery empty messages