dalathegreat/Battery-Emulator #1257: [BUG] rolling reboot after upgrading from 8.12 to 8.14.1

pmajor021, 2025-06-23:
[Archive.zip](https://github.com/user-attachments/files/20860802/Archive.zip)

**Describe the bug**
Tried to upgrade BE over the air, firmware was built on VS Code, but it stuck on "reboot" page.

Connected via USB and on serial monitor I saw that lily is rebooting continously. 

I applied the same version 8.14.1 via USB, but the result was the same, so I reverted to the previous working 8.12.

**To Reproduce**
Upgrade from 8.12 to 8.14.1

**Expected behavior**
Successful upgrade

**Screenshots**
n/a

**Version and settings (please complete the following information):**
- Please attach your USER_SETTINGS files for easier debugging
- Software version: 8.12
- Battery used: BMW i3 (Double battery)
- Inverter communication protocol: BYD_MODBUS
- Hardware used for Battery-Emulator: HW_LILYGO
- CONTACTOR_CONTROL: no
- MQTT: yes

kyberias, 2025-06-23:
This is related to BMW i3 double battery only.

pmajor021, 2025-06-23:
I just tried it and it's the same rolling reboot with version 8.13.

PR #1258 (2025-06-26): Fix BMW I3 double battery crash
### What
This PR fixes #1257. 

### How
A redundant null-pointer access is removed.