dalathegreat/Battery-Emulator #1272: [BUG]

Markus1986, 2025-06-30:
**Describe the bug**
Hello, my RJXZS BMS Battery System with 192 cells via Lilygo on a GEN24 is sometimes not workin properly when charging. Sometimes ist works out fine, but sometimes the Charging power jumps from zero to max. available power an back to zero within seconds and does this for several hours. 

**To Reproduce**
even after 3 days of debugging i can not say how to reproduce the bug.

**Version and settings (please complete the following information):**
// Please attach your USER_SETTINGS files for easier debugging
// Software version: X.Y.Z
- 
- Battery used: RJXZS
- Inverter communication protocol: BYD_MODBUS
- Hardware used for Battery-Emulator: HW_LILYGO
- CONTACTOR_CONTROL: no
- MQTT: no

dalathegreat, 2025-06-30:
The RJXZS BMS is the one who determines charging power. I suggest checking the stats on the BMS via the Smartphone app for the RJXZS, you can post screenshots of both the configuration, and the livedata from the battery. Bonus would be to see the livedata when the charging power goes to 0W

dalathegreat, 2025-07-04:
Wait what, the issues stemmed from you exceeding the battery voltage range? Yeah the smaller Fronius inverters have quite limited voltage ranges. The Datasheets do tell exactly which HVM are supported, and from that you can see how large voltage it expects.

Markus1986, 2025-07-07:
**NO its not the Data for solar panel input** - please read carefully before you lough about me!!!

it is the section for the battery from which the screenshot is.... i marked it yellow - and i think even if its the german version you can understand the word **"BATTERIEanschluss"**

why didnt you tell me that on my first inquiry which i sent you per mail?
That the maximum Voltage is 455V ??