dalathegreat/Battery-Emulator #1274: [BUG] Double Battery Renault ZOE gen1 and GoodWe inverter not working

hedkvist, 2025-06-30:
I wrote about it on Discord, but I thought it would be better to write a Github issue. 
Maybe there is someone who has the same problem, or who has a working setup?
I think i tried everything possible :D
In all exampel i does not change any cabels. 
The only thing i change is Switch 1 and Switch 2 Power from the battery (Se picture.)
And offcourse  downloading the BE files.

When defining like in Exampel 1.(One battery)
(Version 8.13, but i also tried with version 8.14.0, 8.14.1)
CAN1 = Inverter  
CAN2 = Battery 2
Switch 1 OFF
Switch 2 ON
Everything works with one battery

When defining like in Exampel 2.(One battery)
(Version 8.13, but i also tried with version 8.14.0, 8.14.1)
CAN 1 = Inverter AND battery 1.
Switch 1 ON
Switch 2 OFF
Everything works with one battery

When defining like in Exampel 3 (Two battery)
(Version 8.14.1)
CAN1 = Inverter and Battery 1. 
CAN2 = Battery 2. 

There is no RX at CAN 2. 

- Software version: 8.14.1
- Battery used: 2x 41kWh Renault ZOE Gen 1
- Inverter communication protocol: CAN
- Hardware used for Battery-Emulator: HW_STARK
- CONTACTOR_CONTROL: yes
- MQTT: yes

[filer.zip](https://github.com/user-attachments/files/20984164/filer.zip)
[image]
[image]

Maybe i can help somehow? :-)

dalathegreat, 2025-06-30:
Yes, CAN is broken for double support for some reason. I tried to code it exactly as I did with the LEAF support, but maybe I did something wrong. @kyberias was ordering some CAN hardware for testing, let's see if they or me can solve this 😅

kyberias, 2025-06-30:
There is a fix now merged but I didn't yet close this until some is able to test it.

closed 2025-06-30 by kyberias

hedkvist, 2025-07-01:
@kyberias @dalathegreat Unfortunately it dosen't seem to work with this last files. 😭
It actually got worse, I'm not getting any information from any of the batteries or the inverter. 

[image]
[image]

[canlog_08-31-56.txt](https://github.com/user-attachments/files/21005506/canlog_08-31-56.txt)
[USER_SETTINGS.zip](https://github.com/user-attachments/files/21005507/USER_SETTINGS.zip)

dalathegreat, 2025-07-01:
It is probably worse since main is now massively broken, as also seen in https://github.com/dalathegreat/Battery-Emulator/issues/1280

kyberias, 2025-07-02:
There was another fix #1284. Please re-test the latest main.

hedkvist, 2025-07-02:
> There was another fix [#1284](https://github.com/dalathegreat/Battery-Emulator/pull/1284). Please re-test the latest main.

Thank you!

I tried, but when i compile i get:

exit status 1

Compilation error: exit status 1

hedkvist, 2025-07-02:
> There was another fix [#1284](https://github.com/dalathegreat/Battery-Emulator/pull/1284). Please re-test the latest main.

Thanks! Tested, compiling works fine.
Communication to Inverter and battery 1 working, But still no communication to Battery2 on CAN2.

[image]
[image]

[canlog_08-23-50.txt](https://github.com/user-attachments/files/21023039/canlog_08-23-50.txt)
[USER_SETTINGS.zip](https://github.com/user-attachments/files/21023038/USER_SETTINGS.zip)

kyberias, 2025-07-02:
Care to enable DEBUG_VIA_WEB and show the beginning of the log?

hedkvist, 2025-07-02:
> Care to enable DEBUG_VIA_WEB and show the beginning of the log?

Yes offcource. Here it is. :)

[log_09-27-03.txt](https://github.com/user-attachments/files/21023866/log_09-27-03.txt)

hedkvist, 2025-07-15:
@dalathegreat @kyberias 

Did the log give any clues? anything else I can test/do?

kyberias, 2025-07-16:
@hedkvist I should soon get suitable HW to test this with and I can try to reproduce and troubleshoot more.

PR #1276 (2025-06-30): Use the correct CAN interface when Zoe sends in double battery setup
### What
This PR likely fixes #1274 

### How
The current Zoe Gen1 code incorrectly always sent to battery 1 CAN instead of the object's configured CAN interface.