dalathegreat/Battery-Emulator #1280: [BUG] Build from main does not communicate with inverter (Solax Ultra)

freddanastrom, 2025-07-01:
**Describe the bug**
Building the source code from current main disrupts the communication between inverter and emulator.
No visible events or errors in emulator except that main page indicates that inverter does not allow contactor closing.
Inverter gives BMS lost fault.

Reverting back to commit 5650560 (PR 1251) with the same settings makes everything to work as expected, so something in between this has caused this bug.

**To Reproduce**
Build current main

**Expected behavior**
Communication with inverter to work

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version and settings (please complete the following information):**
- Please attach your USER_SETTINGS files for easier debugging

[USER_SETTINGS.cpp.txt](https://github.com/user-attachments/files/21000433/USER_SETTINGS.cpp.txt)
[USER_SETTINGS.h.txt](https://github.com/user-attachments/files/21000434/USER_SETTINGS.h.txt)

- Software version: 8.15-dev
- Battery used: Byd Atto 3
- Inverter communication protocol: Solax CAN
- Hardware used for Battery-Emulator: HW_STARK]
- CONTACTOR_CONTROL: [no]
- MQTT: [yes]

kyberias, 2025-07-02:
The latest main after merging of #1284 probably works better.

freddanastrom, 2025-07-03:
Unfortunately I won’t be able to test this until earliest tomorrow.

closed 2025-07-07 by freddanastrom