dalathegreat/Battery-Emulator #1285: Byd 50kwh install issue

TheTillerMan, 2025-07-02:
Hi all. I was hoping someone could tell me a work around. Trying to install a byd 50kwh battery through a lilygo to a fronius primo gen 24 inverter. I have done this with multiple tesla battery banks but hitting a snag with the byd. Up on start up the contractors close but then open 1 second later. I am getting an under voltage error however I a reading 342v on the BMS. Just wondered if anyone has had this happen and what the work around is. Maybe I have the code wrong somewhere but all looks good from what I can see.

Thanks in advance!

[image]
[image]

dalathegreat, 2025-07-02:
As per the Wiki https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-BYD-Atto-3-%E2%80%90-Yuan-Plus#50kwh-standard-range

_Note: Contactor closing not yet stable on 50kWh batteries, they close and open again shortly. CAN log file wanted from 50kWh vehicle, so we can solve this!_

TheTillerMan, 2025-07-03:
HI Dala,

Sorry didn't hit send on the can logs. 

I have CAN logs for both the 50kwh and 45kwh batteries as the same issue happens to both. 
Is there an email I can forward you the xmt files to as it wont let me upload them here.

Would be great if you can help. 

Thanks for your time

TheTillerMan, 2025-07-04:
The can logs were taken from the cars.

See link attached.

https://drive.google.com/drive/folders/1Ls9dne7NTCUgEUsl2PlMKluunCf8yWpr?usp=sharing

Thanks again.

dalathegreat, 2025-07-11:
Hi, the file is quite small, with just this;

; C:\Users\irvai\OneDrive\Desktop\45kwh pack ign_on.xmt
;   CAN messages saved by PCAN-View v5.3.1.948
;
; Columns descriptions:
; ~~~~~~~~~~~~~~~~~~~~~
; +Message ID
; |            +Cycle time in ms (0=manual)
; |            |  +Data length
; |            |  |  +Frame type
; |            |  |  |  +Message data
; |            |  |  |  |
054h           0  8  D 01h 00h 00h 00h 00h 00h 00h 00h ; Open Contactors
054h           0  8  D 01h 00h 01h 00h 00h 00h 00h 00h ; Close Contactors

Are you saying you found out that message 0x054 is responsible on the smaller batteries for contactor operation?

TheTillerMan, 2025-08-21:
Hi Bill what would you like me to send across?

TheTillerMan, 2025-08-24:
Thanks Billy, 

When compiling sketch I get this error. 

Any Ideas what I'm doing wrong. 

[image]

TheTillerMan, 2025-08-24:
Hi Billy , 

See link attached. 

https://drive.google.com/drive/folders/1RfQYP7-ry9SP1FhqjgvQgfFCPdtzDK7C?usp=sharing

Thanks

bingflash2, 2025-08-24:
swap your bydxxx.cpp file for this one..

https://drive.google.com/file/d/1VIHcyVO8gOpitrjEFXaXlCN3HyDSbFCe/view?usp=sharing

please provide screenshots from BE if its working 

best regards, billy

TheTillerMan, 2025-08-24:
Hi Again Billy, 

Sorry but it is still giving me the same error

bingflash2, 2025-08-24:
your screenshot is compiling v8.16
you send me via drive, v8.14, I modified the bydxxx.cpp from 8.14 to suit v814.
version control is important!

as for the error, you have a missing file from your v8.16 directory...

re-download v8.16 : [](https://github.com/dalathegreat/Battery-Emulator/archive/refs/tags/v8.16.0.zip)
unzip to a new directory (should solve missing file), configure your settings for ev battery type (#define BYD_ATTO_3_BATTERY
), inverter type(#define BYD_MODBUS), inverter communications(.inverter = RS485) wifi, password etc

dont forget to rename USER_SECRETS.TEMPLATE.h to USER_SECRETS.h after editing your lillygo details

then swap this file into the \Battery-Emulator-8.16.0\Software\src\battery directory : [](https://drive.google.com/file/d/1UMg13m3sZ8YHy8ByjPwX2NNbz_qJ-4Z_/view?usp=sharing)

and report back, best regards billy

TheTillerMan, 2025-08-24:
Hi Billy, 

Sorry to be a pain with this one but it's coming up with the errors again. Im not full ball on the software so struggling to work out my issue 

[image]

Here is my code if it's easier for you to look through it. You should be able to edit that file too.

https://drive.google.com/drive/folders/1woXNL-AT0yOcG-xOi5lDd1JXrzBhuvJP?usp=sharing

bingflash2, 2025-08-24:
first update your arduino ide: help menu, check for arduino updates
should update to2.3.6

under the folder icon on the left side there is the boards icon
under the text 'boards manager' into the textbox enter ESP32 & hit enter
at the bottom esp32 by expressif systems will come up, select v3.3 from dropdown box and hit the update button

that will take a bit, report after you have done this

bingflash2, 2025-08-24:
close arduino IDE

open software.ino from ...\Battery-Emulator-8.16.0\Software directory
in arduino ide tools menu -> boards -. "ESP32 Dev module"

delete this file ...BYD-ATTO-3-BATTERY.cpp from directory ...Battery-Emulator-8.16.0\Software\src\battery

rename BYD-ATTO-3-BATTERY-2.cpp in  directory ...Battery-Emulator-8.16.0\Software to BYD-ATTO-3-BATTERY.cpp and move to 
...Battery-Emulator-8.16.0\Software\src\battery

plugin your lillygo board

configure your settings

add // to line 18 of users settings.cpp (comment out inverter uses can, since your rs485)

sketch menu ->verifiy/compile

sketch menu->upload

bingflash2, 2025-08-24:
Just an update, I have personally upgraded my arduino IDE & esp32/expressif library (last version of BE I compiled was 8.13) with the modified  bydxxx.cpp to suit BE 8.16 and it worked fine, I eagerly await your positive feedback. 

Best regard Billy

TheTillerMan, 2025-08-25:
Hi Billy, I had to uninstall then reinstall arduino IDE then do the update as the original software seemed to have a glitch at my end. Everything is now flashed and I will install and test this morning. 

Thankyou greatly for your help with this and I will report back after install this morning

TheTillerMan, 2025-08-25:
Hi Billy. Installed today and so far so good. We had shit weather here in the South of Western Australia so no solar yield but then not is currently discharging via a fronius gen24 inverter. I will comment back after a few successful charge cycles. 

Thanks again for your help on this 

[image]
[image]

freddanastrom, 2025-08-26:
PR #1444 will officially add support for 50 kWh in future releases.

TheTillerMan, 2025-08-30:
Hi all. I’m getting these errors after 3 days of successful use. I have checked wiring and everything seems to check out. Any ideas?

[image]
[image]
[image]
[image]

freddanastrom, 2025-08-30:
Do you have any activity at all if you open the CAN logger page?

The emulator is currently not able to read the contactor status from the battery, it just sends the commands to close and assume they are closed.

bingflash2, 2025-09-01:
if you let it cool down and try again what is happening ?

TheTillerMan, 2025-09-01:
Hi Billy, it looks like the issue may be from interference. I had ran the 12v supply cable in the same conduit as the battery can (as this is how it seems to be done in the car). I have separated the cable from the can and it has worked for 48hours so far issue free. Only problem i am having is the inverter stops charging the battery at 94% even tho I have set to 100% charge. 
Not a big problem but it would be nice to get it to 100% like the other systems I have installed

bingflash2, 2025-09-01:
the lillgo has an issue on 12v i believe, that can cause it to drop out, run it from 5v, check that on the wiki
if the bms deems its charged, it will override what you ask for, if its a issue, scale your SOC, so both sides are happy
please read the following [https://batteryuniversity.com/article/bu-808-how-to-prolong-lithium-based-batteries](https://batteryuniversity.com/article/bu-808-how-to-prolong-lithium-based-batteries) the leading cause of battery degradation is overcharging, you will triple the life of your battery if you go to 80%

very few people understand the negative impact of overcharging, you can achieve 6000 cycles if you stay within 30-80%.

best regards, billy