dalathegreat/Battery-Emulator #1297: Add support for non-european HV Battery-Storage-System TrinaBESS

TheGreenJosip, 2025-07-08:
Expected Behavior

I have required an HV Li-Ion based Storage System named TrinaHome T-Series N6052T.
The main control unit comes with a battery already attached to it rated at 200V and ~6kWh. Individual packs can be added with 100V and ~3kWh. The cluster communicates over CAN 2.0.

BCU and BMS are from a Chinese company called BMSer - Website: [https://en.bmser.com/product_details/5

I was able to gain CAN-communication and flashed the LilyGo using `TEST_FAKE_BATTERY` option including `DEBUG_CAN_DATA` setup. The LilyGo was attached over USB to my MacBook and I piped the CAN-logging to a file which you'll find attached.

I would greatly appreciate If we could manage to gain proper CAN integration for the respective Battery-Storage.
I thank you in advance for your help and support so far and am happy to provide anything necessary! 👍🏼

[trina-bess-can.zip](https://github.com/user-attachments/files/21129711/trina-bess-can.zip)

dalathegreat, 2025-07-09:
Hi,
Can you contact the manufacturer, and ask for a communication CAN datasheet for the battery? That will make development time take ~1day, instead of ~1month :)

dalathegreat, 2025-07-10:
Try eu.sales@trinabess.com ?

dalathegreat, 2025-07-11:
If you can match the CAN log you provided, with screenshots from an app or another system that shows what SOC% and Voltages the battery sits at, we can reverse engineer parts of it before the manufacturer replies. But like I mentioned, without the CAN document, it will take months instead of a single day of testing.

closed 2026-03-15 by dalathegreat