dalathegreat/Battery-Emulator #1303: [BUG] PYLON-CAN.ccp edit for SOLXPOW X3 20 kW

dvbhannes, 2025-07-12:
According to the WIKI for the SOLXPOW X3 20 kW, the PYLON-CAN.ccp file needs to be modified. Where exactly do I need to add the change at the beginning?

CAN_frame PYLON_7330 = {.FD = false, .ext_ID = true, .DLC = 8, .ID = 0x7330 .data = {0x58, 0x58, 0x58, 0x58, 0x58, 0x58, 0x58, 0x58}}; CAN_frame PYLON_7340 = {.FD = false, .ext_ID = true, .DLC = 8, .ID = 0x7340 .data = {0x58, 0x58, 0x58, 0x58, 0x58, 0x58, 0x58, 0x58}};

Furthermore,

void send_setup_info() { //Ensemble information #ifdef SEND_0 transmit_can_frame(&PYLON_7310, can_config.inverter); transmit_can_frame(&PYLON_7320, can_config.inverter); transmit_can_frame(&PYLON_7330, can_config.inverter); transmit_can_frame(&PYLON_7340, can_config.inverter); #endif

needs to be added. For this, does the section need to be adjusted from lines 320 to 329, or can it simply be added?
void PylonInverter::send_setup_info() {  //Ensemble information
#ifdef SEND_0
  transmit_can_frame(&PYLON_7310, can_config.inverter);
  transmit_can_frame(&PYLON_7320, can_config.inverter);
#endif
#ifdef SEND_1
  transmit_can_frame(&PYLON_7311, can_config.inverter);
  transmit_can_frame(&PYLON_7321, can_config.inverter);
#endif
}
Or can someone provide me with the modified PYLON-CAN.ccp file?

Thanks!

dalathegreat, 2025-07-12:
[PYLON-CAN.cpp.zip](https://github.com/user-attachments/files/21199961/PYLON-CAN.cpp.zip)

Feel free to use this file. 

Let's keep this issue open, and add a proper configuration option for the SOLXPOW

dvbhannes, 2025-07-13:
> [Verfolgungsbild][https://github.com/notifications/beacon/A5QYALEELGPG2XM36GULRKL3IF4CBA5CNFSM6AAAAACBLTXDQCWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTVWYBDFI.gif]

Thanks Dala! Always great. Once I get the SOLXPOW X3 20 kW running with the Tesla battery, I'll send you some photos of my setup!

Best regads
Henrik

dalathegreat, 2025-07-13:
Even better, I added this as a complete feature in pull request https://github.com/dalathegreat/Battery-Emulator/pull/1307

You can test it out, by downloading it here: https://github.com/dalathegreat/Battery-Emulator/archive/refs/heads/feature/solxpow-protocol.zip

closed 2025-07-21 by dalathegreat

dalathegreat, 2025-07-27:
Hi @dvbhannes 

You need to select the Minimal SPIFFS like  the quickstart video shows how to do:

The Arduino board should be set to ESP32 Dev Module and Partition Scheme to Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS) (under Tools -> Board -> ESP32 Arduino) with the following settings:

[image]

dvbhannes, 2025-07-28:
Thanks,
my PowMr SolxPow X3 20kW is now working reliably. Unfortunately, it still seems to have an issue with the Modbus connection, but this doesn't have any negative impact on the battery emulator. I'm including the 20kW in my inverters. However, there's another identical model under the Solinteg label, MHT 10-20KW. Here's the link to Solinteg: https://www.solinteg.com/product/integ-m-10-20kw/