dalathegreat/Battery-Emulator #1319: Tesla Model 3 LFP battery contactor not closing

tomilein85, 2025-07-18:
Hello,

I have a Tesla 1666969-24-E Battery (Tesla Model 3 LFP battery) with a Fronius Symo Gen24.
We are already running 3 other Tesla Model 3 batteries and have wired all of them in the same way (jumpering interlocks, suppling 12V via power supply, add capacitor for motor emulation, replacing pyro-fuse).

The LFP pack does not allow closing the contactors and we get the following codes via debug information:
ERROR: BMS_a071_SW_SM_TransCon_Not_Met
ERROR: BMS_a083_SW_Ctr_Close_Failure

The liliygo board with the current Firmware (8.15.0) does detect that it is a LFP chemistry and the battery has a SoC of 80%.

Settings on all of our 4 battery emulator boards are the same:
#define TESLA_MODEL_3Y_BATTERY
#define BYD_MODBUS 

What could we do, to fix the issue?

Macrisu, 2025-07-19:
`/#define LFP_CHEMISTRY          //Tesla specific setting, enable this line to startup in LFP mode`

tomilein85, 2025-07-30:
The Change in the Tesla-battery.h fixed the problem.
Thanks a lot.

closed 2025-07-30 by tomilein85

mariusz34504436, 2025-10-01:
Hello.

I'm from Poland and apologize for my English.

I'm connecting the LFP 2023 without an inverter, but the contactors aren't engaging. The LILIGO v9 software shows:

NEGATIVE CONNECTOR: OPEN

POSITIVE CONNECTOR: PRECHARGE

Could this be caused by the lack of a capacitor on the high-voltage output? The battery is supplying 12.5 volts to penthause.