dalathegreat/Battery-Emulator #1386: [Feature request] JK-BMS

poolcontrol007-git, 2025-08-01:
Hi, 
I have a SMA Tripower STP6.0-3SE40 which of course communicates over CAN to the battery. 
But, the battery packs from XIYIANG have a JK-BMS. they also communicate over CAN, but if I understand correctly it also communicates over RS485. That would fit the Lilygo hardware. Unfortunately I can't choose this BMS in the ESP project. 
Do you have a version where I can select this BMS?

BTW there are 2 Lilygo's in this https://youtu.be/YBYWBapnnyM?feature=shared which are communicating over RS485. Which settings for each Lilygo should be used to configure this setup?

This could be a solution in connecting the SMA(CAN) to the JK-BMS(CAN) instead of JK-BMS(RS485)

Many Thanks,
Luc

bingflash2, 2025-08-13:
The OP, has a XIYIANG LV battery much like this link : https://www.aliexpress.com/p/tesla-landing/index.html?scenario=c_ppc_item_bridge&productId=1005009309664573

the Xiyiang LV battery uses a topbms/cleverbms/JK-BMS which has its own native protocol @ 250kbits/s over CAN.

the protocol documentation, can be found here : https://www.topbms.com/static/upload/file/20230804/1691122017124559.pdf

dalathegreat, 2025-08-16:
@Macrisu This project supports both HV and LV batteries and protocols. Many use 48V systems.

Like @bingflash2 pointed out, OP has a HV inverter, so it makes no sense to ask about LV protocols.

For OP to get the battery going, a custom BMS needs to be added, or some box that combines all LV comms into one single BMS

closed 2025-08-16 by dalathegreat