dalathegreat/Battery-Emulator #1387: [BUG] Homeassistant complains about missing attributes (via MQTT)

kyberias, 2025-08-03:
**Describe the bug**
When battery data is sent via MQTT to Home assistant it fills its log with complaits about some missing values:

```
Template variable warning: 'dict object' has no attribute 'cell_min_voltage' when rendering '{{ value_json.cell_min_voltage }}'
Template variable warning: 'dict object' has no attribute 'cell_max_voltage' when rendering '{{ value_json.cell_max_voltage }}'
Template variable warning: 'dict object' has no attribute 'discharged_energy' when rendering '{{ value_json.discharged_energy }}'
Template variable warning: 'dict object' has no attribute 'charged_energy' when rendering '{{ value_json.charged_energy }}'
Template variable warning: 'dict object' has no attribute 'cell_voltage_delta' when rendering '{{ value_json.cell_voltage_delta }}'

```

**Expected behavior**
We expect BE to configure HA sensors so that these optional (?) missing values are not a problem.

**Version and settings (please complete the following information):**
- Please attach your USER_SETTINGS files for easier debugging
- Software version: 8.14 +
- Battery used: NISSAN_LEAF
- Inverter communication protocol: BYD_MODBUS
- Hardware used for Battery-Emulator: HW_LILYGO
- CONTACTOR_CONTROL: [yes/no]
- MQTT: yes

kyberias, 2025-08-03:
This might be related to the fact that these are values that the HA sends conditionally.

For example:

```
  if (supports_charged) {
    if (datalayer.battery.status.total_charged_battery_Wh != 0 &&
        datalayer.battery.status.total_discharged_battery_Wh != 0) {
      doc["charged_energy" + suffix] = ((float)datalayer.battery.status.total_charged_battery_Wh);
      doc["discharged_energy" + suffix] = ((float)datalayer.battery.status.total_discharged_battery_Wh);
    }
  }

```

kyberias, 2025-08-03:
Here for example while the BE doesn't yet have the cell voltages available, it still reports status to MQTT/HA with these missing values:

```
  if (battery.info.number_of_cells != 0u && battery.status.cell_voltages_mV[battery.info.number_of_cells - 1] != 0u) {
    doc["cell_max_voltage" + suffix] = ((float)battery.status.cell_max_voltage_mV) / 1000.0;
    doc["cell_min_voltage" + suffix] = ((float)battery.status.cell_min_voltage_mV) / 1000.0;
    doc["cell_voltage_delta" + suffix] =
        ((float)battery.status.cell_max_voltage_mV) - ((float)battery.status.cell_min_voltage_mV);
  }

```

kyberias, 2025-08-03:
BE currently configures all the HA sensors (including these in question) with the availability pointing to the same global topic:

```
"availability": [
    {
      "topic": "BE/status"
    }
  ],

```

I guess one possible fix would be to define own availability topics for these sensors so that BE could report them as unavailable while they actually are unavailable.

kyberias, 2025-08-03:
I believe it's configuring each individual sensor separately like so:

[image]

kyberias, 2025-08-03:
I believe we merely need to config these sensors to point their availability to a sensor-specific topic instead of the global "BE/status".

No-Signal, 2025-08-03:
Does your setup have supports_charged equal to true?

kyberias, 2025-08-03:
My setup doesn't have, that's a good point. At least for those values it shouldn't even register those sensors then.

No-Signal, 2025-08-03:
I think there may be 2 issues then, the first being values that may be delayed in sending and the second that the sensors shouldn't be configured. As the messages are retained incorrect messages need to be deleted to resolve the issue.

kyberias, 2025-08-04:
So it might actually make sense for BE to clear up the retained sensor configs if the setup doesn't support them in case an earlier (incorrect) config created them.