dalathegreat/Battery-Emulator #1429: [BUG]  battery overheat--charge_limit_exceeded- battery_temp_deviation_high.....

modellfischen, 2025-08-17:
Hello Dala,

I have had the following problem for about a week:
Our Nissan battery with the Gen 24 inverter has been running very reliably in our basement for about a year.
Now I have closed the housing in which the Lilygo board was installed. At the same time, the battery was coverd in sand for fire safety reasons.
Since then, I have had the following problems:

I always get these error messages after about 2 hours, regardless of whether the battery is charging or discharging. (the cpu temp messagge is comming after a view minutes)
After a reboot, it runs again for 2 hours.

I have had difficultys getting the temperature of the Lilygo below 80 degrees with coolers and fans, even though I have reopened the housing.

The temperature of the battery is constant at 16-17 degrees.

What could be causing these obviously false error messages?
I am now on V8.16.

Best regards, Max

[image]
[image]

- Software version: 8.16 and 8.14
- Battery used: NISSAN_LEAF 27kwh
- Inverter communication protocol: BYD_MODBUS
- Hardware used for Battery-Emulator: HW_LILYGO
- CONTACTOR_CONTROL: yes
- MQTT: no

[USER_SETTINGS.cpp](https://github.com/user-attachments/files/21821005/USER_SETTINGS.cpp)

/no]

dalathegreat, 2025-08-17:
Hi,
I would not worry about the Battery-Emulator LilyGo CPU temperature. We will change how it behaves for the 9.0 release.

HOWEVER, the LEAF battery temperature is worrying. It looks like the battery is actually having issues with one of the temperature measurements, as evident by the:
- CHG stop request (very rare message)
- Battery overheat (also quite rare)
- Temp deviation high

Based on this, it sounds like one of the 4 sensors is not reading correctly, it might be enough for a short peak of glitched reading for the BMS to notice this. Since the LEAF BMS raises the CHG STOP REQUEST, it means the measurement is actually broken, and this is not CAN corruption between the LilyGo-Battery

Sorry to be the bearer of bad news :( This is very rare failure. I suggest logging the temperature min/max readings via MQTT to gain more insight into how frequent this failure occurs, and if we get any glitched readings sampled by the emulator.

You can also Leafspy, or soon the "More battery info" page to see the raw measurements coming from the BMS. I can open a PR with adding this functionality to the Battery-Emulator, and ping you there when it is ready for testing.

modellfischen, 2025-08-17:
wow thank you
how fast was this awnser?
i will try this out wright now!

modellfischen, 2025-08-17:
[image]
[image]
[image]

that seems to be not realistic

closed 2025-08-30 by dalathegreat