dalathegreat/Battery-Emulator #1432: [BUG] OUTPUT 1 Stark CMR starts oscillating after precharge time-out

Denulle, 2025-08-17:
**Describe the bug**
The BE correctly starts the automatic pre-charge sequence with the HIA4V1 board but when the correct pre-charge voltage is not reached within the defined time period the BE starts rapidly switching Output 1 on and off as it stops the pre-charge sequence and then tries to restart it again. In order to prevent the relay from oscillating (and breaking over time) a time-out could be implemented so it stops initiating the pre-charge sequence after x tries. A full power cycle of the BMS is necessary before reattempting the pre-charge procedure after a time-out has occurred.

**To Reproduce**
Disconnect the PWM signal for the HIA4V1 so it cannot complete the pre-charge sequence or if manually pre-charging, don't adjust the voltage. Let it time-out and Output 1 starts oscillating after a couple of seconds.

**Expected behavior**
The BE should only try the pre-charge sequence a couple of times if it failed and then stop completely after a couple attempts. 

**Screenshots**
[log_00-00-29.txt](https://github.com/user-attachments/files/21822363/log_00-00-29.txt)

**Version and settings (please complete the following information):**

- Software version: V8.16.0
- Battery used: MEB
- Inverter communication protocol: CAN (separate CAN channel)
- Hardware used for Battery-Emulator:  HW_STARK
- PRECHARGE_CONTROL: yes
- CAN_ADDON: no (native CANFD Stark)
- WEBSERVER: yes
- MQTT: no

dalathegreat, 2025-08-17:
Potentially Stark V1 / V2 incompatibility, needs investigation if this only affects V1

dalathegreat, 2025-08-19:
I have implemented a fix in PR https://github.com/dalathegreat/Battery-Emulator/pull/1439

Direct download link for testing: https://github.com/dalathegreat/Battery-Emulator/archive/refs/heads/bugfix/precharge-osc.zip

PR #1439 (2025-08-30): Precharge: Avoid oscillation in precharge pin incase it fails/timeouts
### What
This PR fixes oscillating precharge pins in  the event of precharge failure

### Why
Fixes #1432

### How
We now return early incase the code has ever ended up in AUTO_PRECHARGE_FAILURE state

If logging is enabled, we send this message "Precharge: CRITICAL FAILURE (timeout/BMS fault) -> REQUIRES REBOOT"

The Event will also be set; Automatic precharge FAILURE. Failed to reach target voltage or BMS timeout. Reboot emulator to retry!
closed 2025-08-30 by dalathegreat