dalathegreat/Battery-Emulator #1458: [BUG] can playback interface names are either confusing or not working

balaur, 2025-08-25:
**Describe the bug**
When sending a replay - eg to test mcp2515 setup - regardless of which side the transmission happens (lilygo's can), extension's can - they show up as TX1.

**To Reproduce**
1) Set a test replay
```
% cat can_dbg_tx0.txt 
(7.556) TX0 54A [8] 10 0 70 2 0 0 0 2B 
(7.558) TX0 54B [8] 1 8 80 12 4 0 0 0 
```
2) Replay it from CAN Native and then replay it from CAN Addon

**Expected behavior**
Expect to see different TX names - one for Native and one from Addon

**Screenshots**

[image]

**Version and settings (please complete the following information):**
- Software version: Software: 9.0.RC1experimental Hardware: LilyGo T-CAN485

dalathegreat, 2025-08-25:
Good improvement suggestion. We used TX0 and RX1 to get compatibility with SavvyCAN format. We basically bundle up every single transmit interface as TX0, and every single incoming interface as RX1. This is indeed confusing.

When we fix this, careful testing that playback and savvycan support still works is needed!

jonny5532, 2025-09-02:
My local fix for this is to use the the internal BE interface number multiplied by two, so you get:

native can: RX0 TX1
native canfd: RX2 TX3
mcp2515: RX4 TX5
mcp2518: RX6 TX7

I haven't actually tested in SavvyCAN but I assume it'd work as expected...