dalathegreat/Battery-Emulator #1536: Hey Dala I need your help!

Caseyw81, 2025-09-13:
Hey I have been following this project for the past year and have finally put together a solar system which I just got running.
What I need help with now is getting the final settings just right to make it work together correctly. I'm not good with CAN, or Lilly-go stuff but you guys have shown me enough to get started.  The last thing I want is to assume everything is fine and my system and or battery get damaged.  So yea I need some Pro help with getting it just right.

[image]
[image]

What I have so far is a 10kw Solar system with Two Solis 11.4 Split phase inverters and a Tesla LFP battery pack.
I'm using a Lilly-go currently but would like to upgrade to a stark in the future if needed.

The trouble I'm having now that the HVB contactors have closed and the Inverter is communicating with LG is the final settings.  I want to make sure everything is correct with the Solis Inverter and LG to safely commission the system and use it for off-grid on my house.  I know you guys are having issues with the Tesla LFP battery pack not balancing because of a CAN signal not known?  I can probably help solve this issue as I have a second pack ready to use but its still in the donor car.  With a few repairs I can get this car charging and possible record the needed CAN data to help solve the balancing issue.  

Let me know if you have time to get me some support and I don't mind paying for your time.

Thanks and have a good one,
Casey

Caseyw81, 2025-09-13:
Here is a link to my Youtube channel showing the first startup. 

https://youtube.com/shorts/unnRd2NChfo?si=ewS9vr3H4I-tUQ_G

closed 2025-09-14 by Caseyw81

Mattski81, 2026-09-02:
Hey Casey,
Did you manage to get the split phase inverters to share the one battery?

Caseyw81, 2026-09-03:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

Hey,
No I ended up running two separate Tesla LFP packs for the inverters.  They
could be powered both by one but would not communicate at the same time.
There was also talk of inverter isolation testing that caused damage to
inverters.
I didn't want to risk that as this is obviously not the intended
installation.
It's been 8 months now and the system is now running perfectly once all
settings and issues were resolved.

This was my old video showing it hooked to one battery, I'll post one soon
of it all finished with two.
https://youtube.com/shorts/unnRd2NChfo?is=ChSYmdyuV2rd8HDj