dalathegreat/Battery-Emulator #1556: Check if it is possible to replace combiner box from Dyness

arjenhiemstra, 2025-09-21:
**Background:**
I have two (yes too expensive ;-)) Dyness Towers connected in parallel, my deye 15k has 2 BMS ports so I thought to make the HV DC side parallel combiner box myself and use the second BMS port for the second battery. According to Deye this should work but it doesn't, only one battery gets seen correctly by the Deye firmware.

Dyness sells a way to expensive combiner box with a piece of hardware that receives CAN signals from the individual batteries and reports them to the inverter as 1 battery. And digging around is how I got here on this great repo!

**Now the actual check question;** 
If I connect the 2 Dyness batteries to 1 CAN port of a Lilygo 2CAN board and my Deye inverter to the other CAN port, sssuming the Dyness talks somewhat default Pylon protocol, it seems that it should be possible to use this project (or I put in some effort to extend it with Dyness battery support) to solve this "combine 2 dyness HV batteries and report them as 1 to the Deye inverter" problem. Correct?
I saw support for double batteries on the wiki and also the Deye inverter seems to be supported. Almost seems to good to be true! Great work!

Anything that I'm missing?

fastbike, 2025-09-22:
> Hello! Nice to find the name Dyness here :-D.
> 
> I'm asking myself if it would be possible to use a Dyness Tower Battery with a Fronius Gen24 with the battery emulator, instead of the expensive BYD batteries...
> 
> I would be very happy about a feedback?
> 
> Regards, Daniel.

Given the official list of supported inverters includes many already supported by BE, I'd hazard a guess that this is possible. E.g I use a Solis S6-EG3P10K-H which is in the Dyness list - we set this up in the inverter as a BYD battery.

jopeyyyy, 2025-09-22:
I have another solar system at work with the Solis S6-EG3P15K-H and the dyness tower T10. That works out of the box as the Solis S6 is compatible to Dyness Tower batteries. I don't remember what I configured in the inverter... 

At home, I have a Fronius Symo Gen24 10kw + symo 10kw and I wanted to add a battery. But the Fronius Symo Gen24 only supports LG and BYD Hvm which are expensive compared to the Dyness Tower. And as I need a big battery, it would be great, if I could use a cheap Dyness Tower T21 (maybe even 2x T21 in the future). 
So I thought to use the BE, to emulate a BYD battery for the Fronius. 
I just don't know if the BE will communicate with the Dyness. From the other side should be no problem as the Fronius Symo Gen24 is well supported by BE... 

Is there any test I could do with the BE+laptop on the running system Solis/Dyness, to check if BE communicates with the Dyness Tower? 

Because if yes, jackpot!

miraxm, 2025-09-22:
I am able to test it. I using and installing DYNESS batteries for last three years. I have experience with combiners too. I have bought one lyligo board. I just never did anything, because I am waiting for support for maxus even batter I harvested from my broken car. My question: is the only one board enough to test the DYNESS battery communication?

jopeyyyy, 2025-09-24:
Great! I think I will order too. But regarding the code adjustment, I think I would be lost.
So I will wait for your test :-)

But thanks a lot and waiting for your testing...

arjenhiemstra, 2025-10-05:
Ok, just received the lilygo and making the first steps. The battery is not correctly supported by any of the current battery protocols of BE but the Pylon protocol gave the best data so far. I started to created the necessary changes to add support for the Dyness HV battery protocol, which will probably take some time to completely reverse engineer. Once it works I'll create a PR.

[image]

jopeyyyy, 2025-10-05:
wow great! Thank you very much for your work! good news for me...

weldingukraine-byte, 2025-12-25:
Hi, could you tell me what the pinout was when connecting the BE to the battery, and has anyone already implemented a Fronius Dyness connection? Thanks for the great work!

tekampe-oss, 2026-01-31:
Hello,

I would also like to have the pinout and, if possible, the battery configuration. With the default Pylon configuration, but had no luck. With BYD Atto 3 I get at least a SoC from the Dyness Tower.