dalathegreat/Battery-Emulator #1578: [BUG]StarkCMR problems with v9.1 and v9.1.1

Macrisu, 2025-09-28:
**Describe the bug**
I have a new StarkCMR v2, never connected to the battery. (I still use lilygo)
I've installed via webpage v9 and was working ok, but v9.1 and v9.11 are not saving the settings (WIFI and Fake battery)

**To Reproduce**
Updated to v9.1 first, OTA = error -> USB to v9.1, but problems with saving settings.
Downgraded to 9.0 via USB
Updated v9->v9.11 via OTA OK (lost all settings), but problems with saving settings on reboot.

**Expected behavior**
Problems:
1. I have 2 SSID
- one with pass _xxxx.xxx345!@#_
- one with pass _xxxxxx2025?_

The SSID is working on save (get's IP), but it is lost on reboot.

2. WIFi Hostname cannot be used with any chars.

3.  SSID for the **BatteryEmulator** contains the mac address

**Screenshots**

[image]

[image]

dalathegreat, 2025-10-07:
Issue: 2 & 3 was fixed in software 9.1.2 release

Issue 1, settings getting corrupted for some Stark boards still happen, under investigation

Macrisu, 2025-10-07:
For me all ok now.
Updated via OTA.